cherry-embedded/CherryUSB #260: USB Host HID class driver fails if the device has a long class descriptor

dkonigsberg, 2024-09-25:
### 描述一下这个bug / Describe the bug

CherryUSB's host class driver for USB HID devices fails to connect if the class descriptor length (usb_hid_descriptor, wDescriptorLength) is greater than 256 bytes.

In the code, the failure is here:
https://github.com/cherry-embedded/CherryUSB/blob/master/class/hid/usbh_hid.c#L217-L222

While the majority of USB HID devices have class descriptors shorter than 256 bytes, some actually do have descriptors longer than 256 bytes.

When doing a little bit of research, it seems like some versions of Windows support class descriptors up to 4k, while it might be possible for them to be as large as 64k.

There may be several ways to solve this issue, as it probably does not make sense to increase the size of `g_hid_buf` or `usbh_hid.report_desc` as they are statically allocated.

One thing I noticed that the HID class driver does not appear to actually parse the class descriptor itself. So it might instead make sense for this request to be optional, or to expect the application code to request the descriptor if necessary and allocate its own buffers for it.

### 复现步骤 / To Reproduce

- Start an embedded device running CherryUSB as a host
- Insert a HID device with a large (>256 byte) class descriptor
- Observe the logs

One device that has this large descriptor problem is the [FTDI FT260 Bridge](https://ftdichip.com/products/ft260q/), which I'm using in a number of projects.  In the configuration I'm using it, as an I2C/GPIO bridge, it has a descriptor that is 279 bytes.

### 设备信息 / Target Device

STM32F446RE, DWC2 port, FreeRTOS

### 日志 / Log

```
[I/usbh_hub] New full-speed device on Bus 0, Hub 2, Port 1 connected
[I/usbh_core] New device found,idVendor:16d0,idProduct:132c,bcdDevice:2200
[I/usbh_core] The device has 1 bNumConfigurations
[I/usbh_core] The device has 2 interfaces
[I/usbh_core] Enumeration success, start loading class driver
[I/usbh_core] Loading hid class driver
[E/usbh_hid] HID report descriptor too large
```

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2024-09-26:
You are right, i will remove report descripor request in stack and called by users. The reason is that we do not parse report desc because it is so complex and need too much memory usage, and in general, when we are working on a usb hid project, we are aware of the device's report information, such as the report id and no need to parse from report desc.

sakumisu, 2024-09-26:
And thank you for your such normative issue information with issue template.

sakumisu, 2024-09-27:
Done

closed 2024-09-27 by sakumisu