cherry-embedded/CherryUSB #265: bug(port/dwc2/usb_dc_dwc2):  removing iso imcomplete isr caused iso transfer to fail

CutClassH, 2024-10-17:
### 描述一下这个bug / Describe the bug

我正在使用cherryusb开发UAC2.0 speak+mic设备。当Endpoint的bInteval != 1时，ISO IN无法正确的传输。
 我猜测是这笔提交导致的。[https://github.com/cherry-embedded/CherryUSB/commit/76bbd09fed8d8781e50c23df273af5e9abe01434](https://github.com/cherry-embedded/CherryUSB/commit/76bbd09fed8d8781e50c23df273af5e9abe01434)
另外 ，原来的代码这里有误：
https://github.com/cherry-embedded/CherryUSB/blob/a8ef0c4cac95a14d76f2c3be9c34c06dd441e7d4/port/dwc2/usb_dc_dwc2.c#L1183-L1203
应该是
```c
daintmask &= 0xFFFF; 
``` 

### 复现步骤 / To Reproduce

audio v2 mic的demo应该可以复现

### 设备信息 / Target Device

USB IP: DWC2
USB Speed：High Speed
Endpoint Decriptor:
[image]

### 日志 / Log

[image]

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2024-10-17:
只能使用interval 1，其余不支持，原来的代码是其他厂家提的，没有用并且也不对，都删了

sakumisu, 2024-10-17:
iso.只能使用 desc dma模式才能使用

CutClassH, 2024-10-17:
hs的spec里面是允许interval 大于1的。
我修改后可以正常传输。之前的版本只能使用interval 1，应该是因为USB_OTG_GINTSTS_IISOIXFR中断处理把out EP当in EP了。
所以我认为这段代码是有用的，只需要fix bug。
```c
- daintmask >>= 16; 
+ daintmask &= 0xFFFF; 
```

CutClassH, 2024-10-17:
但是我不确定在其他厂商的芯片上是否有效，可能需要在带HS的dwc2的芯片上试试

sakumisu, 2024-10-17:
推荐放弃，因为这个中断频率非常高，他的意思是你没有及时填充数据，建议放弃该中断

sakumisu, 2024-10-17:
你的传输长度不允许超过一个 ep_mps，重复调用  usbd_ep_write 也是一样的效果

sakumisu, 2024-10-17:
反正坑点很多，具体细节就不多赘述了

CutClassH, 2024-10-17:
使用dma的话只是每125us在中断里面翻转一次SEVN ODD Frame的标志位吧，应该是可接受的。
如果是Interval只能为1，意味着上层每个125us就要处理一次IN/OUTbuffer，感觉开销更大一些

sakumisu, 2024-10-17:
> 使用dma的话只是每125us在中断里面翻转一次SEVN ODD Frame的标志位吧，应该是可接受的。 如果是Interval只能为1，意味着上层每个125us就要处理一次IN/OUTbuffer，感觉开销更大一些

是的，所以需要用 desc 模式，我这边厂家的客户要求中断频率不允许高，起码5ms以上，所以这种做法已经放弃了。

sakumisu, 2024-10-17:
重复调用 usbd_ep_write 也是有切换的作用的，这个IP 在不使用 desc的时候，iso 需要设置成一个包的大小。其他协议栈比如 ST 自己的或者 USBX 都是不对 iso 未完成中断处理的。

CutClassH, 2024-10-17:
好的，我先看看st是怎么实现的

sakumisu, 2024-10-17:
https://github.com/STMicroelectronics/stm32-mw-usbx/blob/main/common/usbx_stm32_device_controllers/ux_dcd_stm32_callback.c

sakumisu, 2024-10-17:
st 自己的则是没有做处理，因为是在外部重新调用

[image]

sakumisu, 2024-10-17:
剩下的我就不说了，如果你觉得可行，就按照你的方式来

CutClassH, 2024-10-17:
感谢。我感觉在我手上这个芯片平台上，这样做应该是效率比较好的了。不过确实不适用于其他dwc2 ip的芯片。

sakumisu, 2024-10-20:
暂时关闭

closed 2024-10-20 by sakumisu