cherry-embedded/CherryUSB #270: 1.4.1版本 bl616/bl618 无法使用 msc

Dozingfiretruck, 2024-10-30:
### 描述一下这个bug / Describe the bug

我想使用msc，提示需要格式化，点击格式化会失败
cdc功能验证正常

### 复现步骤 / To Reproduce

最新1.4.1使用 msc_ram_template.c做测试

### 设备信息 / Target Device

bl618

### 日志 / Log

无异常日志输出

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2024-10-30:
测试没有问题，建议再自己研究下。使用我们官方 demo 下的template

Dozingfiretruck, 2024-10-30:
> 测试没有问题，建议再自己研究下。使用我们官方 demo 下的template

你好，官方demo下的我也测试了是一样的，

https://github.com/Dozingfiretruck/bouffalo_sdk

这是我的代码，bouffalo部分是直接拉的子模块，跑 usbd_msc_ram 就能复现

sakumisu, 2024-10-30:
https://github.com/cherry-embedded/CherryUSB/blob/master/demo/msc_ram_template.c 这才是我们官方，请注意

Dozingfiretruck, 2024-10-30:
> https://github.com/cherry-embedded/CherryUSB/blob/master/demo/msc_ram_template.c 这才是我们官方，请注意

这个我刚才跑了，是一样的效果，和上面发的是一样的效果，都是格式化失败

sakumisu, 2024-10-30:
那就不知道了，自己解决把

Dozingfiretruck, 2024-10-30:
> 那就不知道了，自己解决把

你这边测试正常的是么？那我再看看，我这测试1.0.0是正常的

sakumisu, 2024-10-30:
撤到1.4再看看呢

Dozingfiretruck, 2024-10-30:
> 撤到1.4再看看呢

今天撤到过1.4.0测试，情况一样，也是格式化失败，但是cdc功能都是正常的，之后没有继续回退

sakumisu, 2024-10-30:
那就没办法了，应该是你移植问题了。参考 https://github.com/hpmicro/hpm_sdk/tree/main/middleware/cherryusb

Dozingfiretruck, 2024-10-30:
> 那就没办法了，应该是你移植问题了。参考 https://github.com/hpmicro/hpm_sdk/tree/main/middleware/cherryusb

实际上移植文件我直接用的官方的 https://github.com/cherry-embedded/CherryUSB/blob/master/port/bouffalolab/usb_dc_bl.c 没有改动任何代码，我再看下是否有问题，谢谢

sakumisu, 2024-10-30:
下载这个测试下 [cherryusb_bl616.zip](https://github.com/user-attachments/files/17570053/cherryusb_bl616.zip)

Dozingfiretruck, 2024-10-30:
> 下载这个测试下 [cherryusb_bl616.zip](https://github.com/user-attachments/files/17570053/cherryusb_bl616.zip)，终端输入 msc

[image]
好像这个命令没注册？

sakumisu, 2024-10-30:
[cherryusb_bl616.zip](https://github.com/user-attachments/files/17570234/cherryusb_bl616.zip)

sakumisu, 2024-10-30:
还有问题的话，加群把

Dozingfiretruck, 2024-10-30:
> [cherryusb_bl616.zip](https://github.com/user-attachments/files/17570234/cherryusb_bl616.zip)

这个能正常格式化

sakumisu, 2024-10-30:
那就检查移植问题把

Dozingfiretruck, 2024-10-30:
> 那就检查移植问题把

OK。非常感谢

sakumisu, 2024-10-30:
[image]

sakumisu, 2024-10-30:
[usb_config.zip](https://github.com/user-attachments/files/17570365/usb_config.zip)

sakumisu, 2024-10-30:
[image]

Dozingfiretruck, 2024-10-30:
> [usb_config.zip](https://github.com/user-attachments/files/17570365/usb_config.zip)

问题找到了，我没有在usb_config.h 中定义 CONFIG_USB_HS ，但是这个是sdk默认定义的。。。 
[image]

sakumisu, 2024-10-30:
有没有一种可能，本来人家sdk里cmake里是有的，现在你用了我这边的cmake。他没了

Dozingfiretruck, 2024-10-30:
> 有没有一种可能，本来人家sdk里cmake里是有的，现在你用了我这边的cmake。他没了

是这样的，sdk中定义的CONFIG_USB_HS传给cmake，原版sdk中在cherryusb中是自己sdk写的cmake编译的，其中加了这个宏，哈哈哈，问题解决了，感谢感谢

closed 2024-10-30 by Dozingfiretruck