cherry-embedded/CherryUSB #276: dwc2_reset 方法在HS上需要修改手动清除RST状态

Clearhu, 2024-11-24:
### 描述一下这个bug / Describe the bug

我在一款测试芯片上，发现 USB_OTG_GLB->GRSTCTL |= USB_OTG_GRSTCTL_CSRST 后该位一直保持为1，并没有自动清除

### 复现步骤 / To Reproduce

内部测试芯片所以无法给出复现方法，可能其他芯片会遇到

### 设备信息 / Target Device

内部测试芯片所以无法给出复现方法

### 日志 / Log

错误的日志：
00> GRSTCTL111: 80000000, 40040010
00> GRSTCTL222: A0000001, 40040010
00> GRSTCTL333: A0000001, 40040010

正确的日志：
00> GRSTCTRL111: 80000000, 40040010
00> GRSTCTRL222: 80000001, 40040010
00> GRSTCTRL333: 80000000, 40040010

### 配置 / Configuration

原始代码为：
    do {
        if (++count > 200000U) {
            break;
        }
    } while ((USB_OTG_GLB->GRSTCTL & USB_OTG_GRSTCTL_CSRST) == USB_OTG_GRSTCTL_CSRST);

修改后位：
    do {
        if (++count > 200000U) {
            break;
        }
    } while ((USB_OTG_GLB->GRSTCTL & USB_OTG_GRSTCTL_CSRST) == USB_OTG_GRSTCTL_CSRST);
    USB_OTG_GLB->GRSTCTL &= ~(USB_OTG_GRSTCTL_CSRST);

### USB中断 / USB Interrupt

不修改前无法进入中断

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2024-11-24:
这个应该是ic那边设计问题，建议找ic设计，目前没有遇到过哪家芯片有这问题

Clearhu, 2024-11-25:
[image]
这个IC确实不会自动清除，看能不能把我这个修改帮忙合进去，也不会影响到其他的ic

sakumisu, 2024-11-25:
以下是IP 手册原文：

`The application can write to this bit any time it wants to reset the
core. This is a self-clearing bit and the core clears this bit after
all the necessary logic is reset in the core, which can take
several clocks, depending on the current state of the core. Once
this bit is cleared software must wait at least 3 PHY clocks
before doing any access to the PHY domain (synchronization
delay).`

Clearhu, 2024-11-25:
[image]
我在ST H7系列的描述看到过这个句话，确实是会自清除的，但是现在我碰到的这个IC没有这么做，并且已经开始量产了，看能不能把这个修改加入进去，再纠结原版这个问题意义也不大，毕竟Cherryusb也是尽可能支持更多IC

sakumisu, 2024-11-25:
可以增加，提PR 把

Clearhu, 2024-11-25:
你帮我弄了算了吧，哈哈

sakumisu, 2024-11-25:
好的

sakumisu, 2024-11-25:
Done

closed 2024-11-25 by sakumisu

sakumisu, 2025-05-12:
自清除功能从 DWC2 IP 4.0 以上开始删除了，改成检测 bit29了