cherry-embedded/CherryUSB #282: GD32F350的USB IP应该是dwc2而不是fsdev吧?

tomzbj, 2024-12-25:
### 描述一下这个bug / Describe the bug

fsdev这边的readme里提到了GD32F350, 而dwc2的readme里没有。

### 复现步骤 / To Reproduce

用fsdev配置，没反应，用dwc2配置则可以正常枚举出usb串口设备.

### 设备信息 / Target Device

芯片型号: GD32F350
引脚: PA11&PA12

### 日志 / Log

[32m[I/USB] ========== dwc2 udc params ==========
[0m[32m[I/USB] CID:00001000
[0m[32m[I/USB] GSNPSID:00000000
[0m[32m[I/USB] GHWCFG1:00000000
[0m[32m[I/USB] GHWCFG2:00000000
[0m[32m[I/USB] GHWCFG3:00000000
[0m[32m[I/USB] GHWCFG4:00000000
[0m[32m[I/USB] dwc2 fsphy type:0, hsphy type:0, dma support:0
[0m[32m[I/USB] dwc2 has 1 endpoints and dfifo depth(32-bit words) is 0, default config: 4 endpoints
[0m[32m[I/USB] =================================
[0m[31m[E/USB] dwc2 has less endpoints than config, please check
[0m[32m[I/USB] fifo0 size:0010, offset:0100
[0m[32m[I/USB] fifo1 size:0100, offset:0110
[0m[32m[I/USB] fifo2 size:0010, offset:0210
[0m[32m[I/USB] fifo3 size:0010, offset:0220
[0m[31m[E/USB] Your fifo config is overflow, please check
[0m[31m[E/USB] Your fifo config is larger than 320 , please check

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2024-12-25:
无所谓，懒得写那么多，错就错了吧，尤其是gd

closed 2024-12-25 by sakumisu

tomzbj, 2024-12-26:
国产这几家里gd算是比较靠谱的了吧... 起码能自己把库搞起来, 不是从ST照抄就完事.

sakumisu, 2024-12-26:
> 国产这几家里gd算是比较靠谱的了吧... 起码能自己把库搞起来, 不是从ST照抄就完事.

不是因为这个原因，是 USB IP 问题，不方便使用。配置信息本来是可以读取的，现在隐藏了，他们自己的库里是一个芯片一个芯片的写配置，我们肯定不会这么做的。

tomzbj, 2024-12-26:
有没有兴趣搞搞WCH的MCU? 他家一直主打各种usb周边, 也有几款mcu内置了HS PHY, CH32F205/F207是cortex-m3的, CH585/CH32V305/307是riscv的. 他家比较不爽的是ram偏小, 以及pyocd目前似乎不支持, 只能用他自家的wchlink.

sakumisu, 2024-12-26:
> 有没有兴趣搞搞WCH的MCU? 他家一直主打各种usb周边, 也有几款mcu内置了HS PHY, CH32F205/F207是cortex-m3的, CH585/CH32V305/307是riscv的. 他家比较不爽的是ram偏小, 以及pyocd目前似乎不支持, 只能用他自家的wchlink.

之前老版本已经支持了，现在不再支持，IP 太乱。