cherry-embedded/CherryUSB #283: EP0 Setup ZLP issue

llinjupt, 2024-12-29:
Hi sakumisu,

Should use ep0_data_buf_residue rather than ep0_data_buf_len to decide if needs a zlp packet in usbd_event_ep0_setup_complete_handler().
It looks like a minor coding mistake. The code below is just for your reference:

```
    if ( g_usbd_core[busid].ep0_data_buf_residue  /* I'm not sure if this line is needed, but host might send a 0-wLength SETUP request */
         && (setup->wLength > g_usbd_core[busid].ep0_data_buf_residue ) 
         && (!(g_usbd_core[busid].ep0_data_buf_residue % USB_CTRL_EP_MPS))) {
        g_usbd_core[busid].zlp_flag = true;
        USB_LOG_DBG("EP0 Set zlp\r\n");
    }
```

sakumisu, 2024-12-29:
Not, we just use g_usbd_core[busid].ep0_data_buf_len to record how many bytes we need send and cannot be changed in the whole setup, ep0_data_buf_residue is is variable in the setup  trasnfer.

sakumisu, 2024-12-29:
And g_usbd_core[busid].ep0_data_buf_residue is the same with ep0_data_buf_len in setup code you showed, we just want to record total bytes.

sakumisu, 2024-12-29:
And your code is incorrect when process setup->out data->ack,  we should know how many bytes we received when call usbd_setup_request_handler in usbd_event_ep0_out_complete_handler, if you use g_usbd_core[busid].ep0_data_buf_residue, it is incorrect because it is a variable and decreased to zero.

closed 2024-12-29 by sakumisu

llinjupt, 2024-12-29:
> And g_usbd_core[busid].ep0_data_buf_residue is the same with ep0_data_buf_len in setup code you showed, we just want to record total bytes.

Yeah, I agree. In this case, they are the same.
The line above "usbd_ep_start_write(USB_CONTROL_IN_EP0, g_usbd_core.ep0_data_buf, g_usbd_core.ep0_data_buf_residue)" made me misundersand it. 

Another small case is if ep0_data_buf_len is 0? May it happen.

sakumisu, 2024-12-29:
If host request data setup->wLength will not be zero, so if ep0_data_buf_len is zero, they are the same and does not match this condition.Also if setup->wLength isnot zero, does not match too.

llinjupt, 2024-12-29:
> If host request data setup->wLength will not be zero, ep0_data_buf_len cannot be zero

If so, you can close this report.