cherry-embedded/CherryUSB #292: usbd_event_handler函数里不会执行USBD_EVENT_CONNECTED和USBD_EVENT_DISCONNECTED?

tomzbj, 2025-02-22:
### 描述一下这个bug / Describe the bug

我希望有个简单的办法来判断当前USBD是否连接到PC， 不过在    
usbd_event_handler函数的switch里，
case USBD_EVENT_CONNECTED:和    case USBD_EVENT_DISCONNECTED:这两行后面加了调试语句发现都没有输出.
可能是我的思路不对？

### 复现步骤 / To Reproduce

在每个case后面都加了调试, 发现只有USBD_EVENT_RESET和USBD_EVENT_CONFIGURED在USB插入时会有输出，USB拔出时则没有任何输出。

### 设备信息 / Target Device

MCU: GD32F303CGT6
PIN: PA11&PA12
IP: FSDEV
CLASS: MSC

### 日志 / Log

无

### 商业 / Business

- [ ] 是 / Yes

tomzbj, 2025-02-22:
又试了一下， 直接hook到这个位置，可以了，我试试能否加个计数器，一段时间没有SOF就判断为USB已断开。
感觉有点傻，应该有更好的办法？
```
 if(wIstr & USB_ISTR_SOF) {
    void SOF_Handler(void);
    SOF_Handler();
    USB->ISTR &= (uint16_t)(~USB_ISTR_SOF);
  }
```

sakumisu, 2025-02-22:
文档里有写，有且仅hpm bl系列支持，其余都不支持，需要自行做vbus检测电路

closed 2025-02-22 by sakumisu

sakumisu, 2025-02-22:
> 文档里有写，有且仅hpm bl系列支持，其余都不支持，需要自行做vbus检测电路

指的是这两个 USBIP 本身支持检测，不需要依赖外部电路

tomzbj, 2025-02-22:
上面我判断有无SOF中断这个办法，看上去似乎能用，不过不知道会不会什么时候就不对了？