cherry-embedded/CherryUSB #293: Support selection of bConfigurationValue for devices with bNumConfigurations > 1

gadget-man, 2025-02-25:
I'm hoping to use the CherryUSB USB Host device class to enable use of a Realtek 8152/8153 usb to ethernet adapter on ESP32s3.

I've tried downloading the demo app, however as soon as I connect a device, I get the following error:

`[E/usbh_core] do not support Class:0xff,Subclass:0xff,Protocl:0x00`

 I believe this is because by default CherryUSB is using bConfigurationValue = 1 (vendor specific class), however I want to use bConfigurationValue = 2 which is CDC-ECM.

Is there a way to implement `CONFIG_USB_HOST_ENABLE_ENUM_FILTER_CALLBACK` and enable `enum_filter_cb` when installing the USB Host, so that we can use the 2nd configuration?

gadget-man, 2025-02-25:
Ah, I've realised that I can manually set `config_index = 1;` in usb_core.c to select the second interface. It would be great if this could be linked to a configuration value rather than hard-coded.

Next step - work out how to link the USB Host device to Netif!

sakumisu, 2025-02-26:
Hi, thanks, we will consider how to fix config_index for users. BTW, we support rtl8152 with 0xff class, also we support class with lwip, you can see them in  https://github.com/cherry-embedded/CherryUSB/tree/master/class/vendor/net and https://github.com/cherry-embedded/CherryUSB/blob/master/platform/none/usbh_lwip.c

gadget-man, 2025-02-26:
Thanks, it's great news that there is already a vendor class for rte8152. I've now found it in menuconfig and tried running the basic sample with it installed. It detects the device fine but then returns an error:

```
2025-02-26 09:43:37 [I/USB] Hello CherryUSB!
2025-02-26 09:43:37 [I/USB] cherryusb, version: v1.4.2
2025-02-26 09:43:37 [I/USB] ========== dwc2 hcd params ==========
2025-02-26 09:43:37 [I/USB] CID:00000000
2025-02-26 09:43:37 [I/USB] GSNPSID:4f54400a
2025-02-26 09:43:37 [I/USB] GHWCFG1:00000000
2025-02-26 09:43:37 [I/USB] GHWCFG2:224dd930
2025-02-26 09:43:37 [I/USB] GHWCFG3:00c804b5
2025-02-26 09:43:37 [I/USB] GHWCFG4:d3f0a030
2025-02-26 09:43:37 [I/USB] dwc2 has 8 channels and dfifo depth(32-bit words) is 200
2025-02-26 09:43:37 [I/usbh_hub] New full-speed device on Bus 0, Hub 1, Port 1 connected
2025-02-26 09:43:37 [I/usbh_core] New device found,idVendor:0bda,idProduct:8152,bcdDevice:2000
2025-02-26 09:43:37 [I/usbh_core] The device has 2 bNumConfigurations
2025-02-26 09:43:37 [I/usbh_core] The device has 1 interfaces
2025-02-26 09:43:37 [I/usbh_core] Selecting configuration 1
2025-02-26 09:43:37 [I/usbh_core] Enumeration success, start loading class driver
2025-02-26 09:43:37 [I/usbh_core] Loading rtl8152 class driver
2025-02-26 09:43:37 [I/rtl8152] rtl8152 version 0x0002
2025-02-26 09:43:37 [E/rtl8152] rx_buf_sz is overflow, requested 16384, default is 2048
2025-02-26 09:43:37 [E/usbh_hub] Port 1 enumerate fail
2025-02-26 09:43:38 
```

It appears that `rtl8152_class->rx_buf_sz` is 16384.

gadget-man, 2025-02-26:
> Thanks, it's great news that there is already a vendor class for rte8152. I've now found it in menuconfig and tried running the basic sample with it installed. It detects the device fine but then [...]

Ignore this - I realised I'd updated `CONFIG_USBHOST_RTL8152_ETH_MAX_RX_SIZE` in the wrong file! Setting it to 16384 from 2048 seems to avoid the issue.

closed 2025-02-26 by sakumisu