cherry-embedded/CherryUSB #302: Cant build the cherryusb host_cdc_msc_hid

Makurisan, 2025-04-11:
It is not clear how to include the cherryusb library into the pico-examples.
The timy usb works without any problems.
How to install and configure cherryusb?

sakumisu, 2025-04-11:
See https://github.com/[CherryUSB/pico-examples](https://github.com/CherryUSB/pico-examples)

sakumisu, 2025-04-11:
https://github.com/[CherryUSB/pico-sdk](https://github.com/CherryUSB/pico-sdk)

closed 2025-04-11 by sakumisu

Makurisan, 2025-04-11:
link doesnt exists

sakumisu, 2025-04-11:
Try the link please, cherryusb is not a component of rp sdk

sakumisu, 2025-04-11:
rp supports tinyusb in their sdk so that you can use make host_cdc_msc_hid, if you want to use cherryusb also, you should ask them to support .

Makurisan, 2025-04-11:
I dont want tiny usb. I see now you have changed the pico-sdk to fetch the lib.
now I cloned pico-sdk and pico-examples and if I make "cmake .." I get:

Using FREERTOS_KERNEL_PATH from environment ('/home/qmk/cherryusb/pico-examples/FreeRTOS-Kernel')
CMake Error at freertos/FreeRTOS_Kernel_import.cmake:84 (message):
  Directory '/home/qmk/cherryusb/pico-examples/FreeRTOS-Kernel' not found
Call Stack (most recent call first):
  freertos/CMakeLists.txt:6 (include)

sakumisu, 2025-04-11:
Solve by yourself

sakumisu, 2025-04-11:
rp official tells you how to add freertos. https://github.com/raspberrypi/pico-examples/blob/master/freertos/FreeRTOS_Kernel_import.cmake

Makurisan, 2025-04-11:
I build examples before and now with your fork of pico-examples I get the error with freeRtos.
Why cant you say what must be done. Normally I call git submodules...

CMake Error at usb/cherryusb/host/FreeRTOS_Kernel_import.cmake:84 (message):
  Directory '/home/qmk/cherryusb/pico-examples/FreeRTOS-Kernel' not found

sakumisu, 2025-04-11:
git clone FreeRTOS-Kernel by yourself, you should ask rp official not me.

sakumisu, 2025-04-11:
I have said, rp official tells you.
[image]