cherry-embedded/CherryUSB #311: 1.4.3版本RAM增大5KB

Clearhu, 2025-05-09:
### 描述一下这个bug / Describe the bug

相比1.4.1版本，最新的1.4.3版本 FLASH减小了10kB，RAM增大了5KB，是否可以进一步优化RAM?

### 复现步骤 / To Reproduce

1.4.1版本：
 | Memory region         Used Size  Region Size  %age Used
 |            FLASH:      251996 B       272 KB     90.47%
 |              RAM:       32080 B        32 KB     97.90%

1.4.3版本：
|            FLASH:      262252 B       272 KB     94.16%
 |              RAM:       37464 B        32 KB    114.33%

### 设备信息 / Target Device

板子型号: CH582M

### 日志 / Log

1.4.1版本：
 | Memory region         Used Size  Region Size  %age Used
 |            FLASH:      251996 B       272 KB     90.47%
 |              RAM:       32080 B        32 KB     97.90%

1.4.3版本：
|            FLASH:      262252 B       272 KB     94.16%
 |              RAM:       37464 B        32 KB    114.33%

### 配置 / Configuration

HID

### USB中断 / USB Interrupt

可以

### 缓存 / Cache

有

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2025-05-09:
需要具体列出是哪个文件

sakumisu, 2025-05-09:
我们的更新中并没有修改ram相关的内容，建议重新 check

Clearhu, 2025-05-09:
编译时使用两个不同版本目录：
CherryUSB_v141
CherryUSB_v143

使用一下文件：
/port/ch32/usb_ch58x_dc_usbfs.c
/class/cdc/usbd_cdc_acm.c 
/class/dfu/usbd_dfu.c 
/class/hid/usbd_hid.c 
/core/usbd_core.c

同样代码编译出来就是一下结果：
1.4.1版本：
| Memory region Used Size Region Size %age Used
| FLASH: 251996 B 272 KB 90.47%
| RAM: 32080 B 32 KB 97.90%

1.4.3版本：
| FLASH: 262252 B 272 KB 94.16%
| RAM: 37464 B 32 KB 114.33%

您可以复现下，其他MCU也是项目情况。

sakumisu, 2025-05-09:
还是那句话，需要每个文件的大小，我这边加起来都没5K，建议重新检查
[image]

sakumisu, 2025-05-09:
ps： 上述测试是两个 bus，如果是单bus，ram除以2。

Clearhu, 2025-05-09:
找到问题了：
新版1.4.3  usb_memcpy.h

#define memcpy usb_memcpy
这句话加上去ram就会大5K, 这个是不是需要添加一个宏可以禁止掉？

sakumisu, 2025-05-09:
> 找到问题了： 新版1.4.3 usb_memcpy.h
> 
> #define memcpy usb_memcpy 这句话加上去ram就会大5K, 这个是不是需要添加一个宏可以禁止掉？

谢谢，可以考虑。但是要说明一点，arm的 memcpy是有bug的，所以你推荐应该怎么弄？

sakumisu, 2025-05-09:
usb_memcpy 不是应该增加的是flash吗，怎么变成 ram了

Clearhu, 2025-05-09:
我估计不久的将来我们要见面了，这在用6264开发产品

Clearhu, 2025-05-09:
arm的 memcpy 没关系，默认打开就好了，我现在发现问题的是ch582，是rsic-v的

sakumisu, 2025-05-09:
测试了下，没有变化，ram没有改变

Clearhu, 2025-05-09:
应该是rsic-v的编译器造成，加个宏好了。

Clearhu, 2025-05-09:
这就奇怪了，是因为用的03吗？

sakumisu, 2025-05-09:
risc-v32/riscv-64/riscv-none 三个都测过了，没变

sakumisu, 2025-05-09:
> 做了几个实验，就是沁恒的编译器造成的

大佬牛逼。要不加个好友把，这个issue回复麻烦的很

Clearhu, 2025-05-09:
v: clearhu

closed 2025-05-09 by Clearhu