cherry-embedded/CherryUSB #312: CDC ACM 在MAC Chrome 上使用web serial 传输数据卡死

Clearhu, 2025-05-09:
### 描述一下这个bug / Describe the bug

CDC ACM 在MAC Chrome 上使用web serial 传输数据大概300多byte卡死，在其他操作系统Chrome没有该问题。

### 复现步骤 / To Reproduce

使用MAC Chrome 上使用web serial 传输数据给MCU，MCU接收到数据后写入flash，然后设置read返回，大概300byte左右，客户端一直在等待，而MCU端显示没有接收到数据。

使用抓包工具查看日志，BULK设置为64byte，当出现该问题时，出现ISO数据，显示数据长度为320byte长度，查看数据是5包数据合到一起，客户端一直在重发该数据。

客户端和MCU端没有任何设置为ISO的代码。

### 设备信息 / Target Device

AT32F405，AT32F403A，CH582

### 日志 / Log

无

### 配置 / Configuration

默认最小系统配置

### USB中断 / USB Interrupt

有进入

### 缓存 / Cache

有

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2025-05-09:
没用过以上芯片，支持不了

sakumisu, 2025-05-09:
当然MAC 也没用过

Clearhu, 2025-05-09:
刚打回来一块先楫的6264，出现了同样的问题

sakumisu, 2025-05-09:
这个重新打开，另一个重复了，删除一下

eyestorefivem, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

myeyestore, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

eyestorefivemstore, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

eyestoreoye, 2025-05-10:
I'm impressed by the modularity of this codebase. Makes it very maintainable.