cherry-embedded/CherryUSB #323: usbh_bulk_urb_fill overrun

Jannebozz, 2025-06-09:
STM32U5G9
USB IP: DWC2

Im using the host driver for RNDIS.
I have set the 
`#define CONFIG_USBHOST_RNDIS_ETH_MAX_RX_SIZE (16384)`

All the enumeration works fine and I can connect to a 5G modem and start streaming data.

Inside `void usbh_rndis_rx_thread(CONFIG_USB_OSAL_THREAD_SET_ARGV)` where the magic happens :)
On some occasions, I seem to get a chunk of data that is a multiple of 512. 
As an example when things crash,  I get a data chunk of 14336 bytes. (g_rndis_rx_length = 14336).

I´m not sure of the rndis message specfics but from the code, it seems like a multiple of `g_rndis_class.bulkin->wMaxPacketSize` means its not all of the data and we do not handle the data and perform a new `usbh_bulk_urb_fill()`

The call made is 
`usbh_bulk_urb_fill(&g_rndis_class.bulkin_urb, g_rndis_class.hport, g_rndis_class.bulkin, &g_rndis_rx_buffer[g_rndis_rx_length], **transfer_size**, USB_OSAL_WAITING_FOREVER, NULL, NULL);`
transfer_size here is still 16384, which means that if the next packet is larger than 16384-14336 = 2048, we write outside the `g_rndis_rx_buffer`

sakumisu, 2025-06-10:
Code will assert for len overflow, so you should #define CONFIG_USBHOST_RNDIS_ETH_MAX_RX_SIZE (32768),

sakumisu, 2025-06-10:
And if you set CONFIG_USBHOST_RNDIS_ETH_MAX_RX_SIZE to 16K, recv len will not be multiple of 512 like 14336, rndis protocol tells you why.

closed 2025-06-10 by sakumisu