cherry-embedded/CherryUSB #326: ESP32-S3 CDC_NCM Host

kcsct, 2025-06-26:
Hi

I'm using https://github.com/CherryUSB/cherryusb_esp32 and enabled CDC_NCM host mode in menuconfig.  When I attach an embedded linux device that is configured as cdc_ncm gadget device enumeration succeeds, DHCP starts but times out.  I have tested the cdc_ncm gadget implementation by attaching the embedded device to my linux PC which then receives a DHCP lease from the embedded device so that works.

Below is the log dump of the ESP32.  Am I doing something wrong with the ESP32's setup?  I haven't change anything in the cherryusb_esp32  repo.  My esp-idf version is 5.5.  Can you please help?

(cpu 0): Args okay. Resulting flags 0x40E             
D (763) intr_alloc: Connected src 79 to int 2 (cpu 0)        
D (768) app_start: Starting scheduler on CPU0                
V (772) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (772) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x402
D (772) intr_alloc: Connected src 57 to int 3 (cpu 0)
V (772) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (782) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (782) intr_alloc: Connected src 80 to int 0 (cpu 1)
D (792) app_start: Starting scheduler on CPU1
V (792) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (802) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x402
D (812) intr_alloc: Connected src 58 to int 1 (cpu 1)
I (772) main_task: Started on CPU0
D (822) heap_init: New heap initialised at 0x3fce9710
V (822) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (832) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (832) intr_alloc: Connected src 52 to int 5 (cpu 0)
I (842) main_task: Calling app_main()
[I/USB] Hello CherryUSB!
 2D (842) esp_netif_lwip: LwIP stack has been initialized
D (852) esp_netif_lwip: esp-netif has been successfully initialized
D (862) event: running task for loop 0x3fceacc4
D (862) event: created task for loop 0x3fceacc4
D (862) event: created event loop 0x3fceacc4
V (872) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (872) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (882) intr_alloc: Connected src 38 to int 8 (cpu 0)
[I/USB] cherryusb, version: v1.5.0
[I/USB] ========== dwc2 hcd params ==========
[I/USB] CID:00000000
[I/USB] GSNPSID:4f54400a
[I/USB] GHWCFG1:00000000
[I/USB] GHWCFG2:224dd930
[I/USB] GHWCFG3:00c804b5
[I/USB] GHWCFG4:d3f0a030
[I/USB] dwc2 has 8 channels and dfifo depth(32-bit words) is 200
I (922) main_task: Returned from app_main()
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x15d8
load:0x403c8700,len:0xd24
load:0x403cb700,len:0x2f84
entry 0x403c8924
I (24) boot: ESP-IDF v5.5-dev-3372-g38628f98b9 2nd stage bootloader
I (25) boot: compile time Jun 26 2025 10:20:00
I (25) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (29) boot: efuse block revision: v1.3
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=11820h ( 71712) map
I (101) esp_image: segment 1: paddr=00021848 vaddr=3fc93300 size=04454h ( 17492) load
[... 188 lines cut ...]
D (6562) esp_netif_handlers: esp_netif action has started with netif0x3fced1c0 from event_0
D (6572) esp_netif_lwip: check: remote, if=0x3fced1c0 fn=0x420133a0
D (6572) esp_netif_lwip: esp_netif_start_api 0x3fced1c0
D (6582) esp_netif_lwip: esp_netif_get_hostname esp_netif:0x3fced1c0
D (6582) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (6592) esp_netif_handlers: esp_netif action connected with netif0x3fced1c0 from event_id0
D (6602) esp_netif_lwip: check: remote, if=0x3fced1c0 fn=0x420135c0
D (6602) esp_netif_lwip: esp_netif_up_api esp_netif:0x3fced1c0
D (6612) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (6612) esp_netif_lwip: check: remote, if=0x3fced1c0 fn=0x42012dd8
D (6622) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3fced1c0
D (6632) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3fced1c0
D (6632) esp_netif_lwip: if0x3fced1c0 start ip lost tmr: interval=120
D (6642) esp_netif_lwip: starting dhcp client
D (6642) esp_netif_lwip: call api in lwip: ret=0x0, give sem
3ImuEbu_chccnrm] dr ate sdcpnrm rl shrea0a,S[0class:0x00,Protocl:0x01
D (126632) esp_netif_lwip: esp_netif_ip_lost_timer esp_netif:0x3fced1c0
D (126632) esp_netif_lwip: check: local, if=0x3fced1c0 fn=0x42013914
D (126632) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fced1c0
V (126632) esp_netif_objects: esp_netif_next_unsafe 0x0
V (126642) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3fced1c0
V (126642) esp_netif_objects: esp_netif_next_unsafe 0x3fced1c0
V (126652) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3fced1c0
D (126662) esp_netif_lwip: if0x3fced1c0 ip lost tmr: raise ip lost event
D (126662) event: no handlers have been registered for event IP_EVENT:5 posted to loop 0x34

sakumisu, 2025-06-26:
I don't know about this, you can try cdc rndis instead.

kcsct, 2025-06-26:
I enabled just RNDIS in menuconfig but it is not detected.

    Select usb host ip, and some ip need config in usb_config.h, please check (CUSTOM (Implement it yourself))  --->
[ ] Enable usb cdc acm driver
[ ] Enable usb hid driver
[ ] Enable usb msc driver
[ ] Enable usb cdc ecm driver
[*] Enable usb rndis driver
[ ] Enable usb cdc ncm driver
[ ] Enable usb video driver, it is commercial charge
[ ] Enable usb audio driver, it is commercial charge
[ ] Enable usb bluetooth driver
[ ] Enable usb asix driver
[ ] Enable usb rtl8152 driver
[ ] Enable usb ftdi driver
[ ] Enable usb ch34x driver
[ ] Enable usb cp210x driver
[ ] Enable usb pl2303 driver
[ ] Enable usb aoa driver
[ ] Use usb host template

I (842) main_task: Calling app_main()
[I/USB] Hello CherryUSB!
 2D (842) esp_netif_lwip: LwIP stack has been initialized
D (852) esp_netif_lwip: esp-netif has been successfully initialized
D (862) event: running task for loop 0x3fceacc4
D (862) event: created task for loop 0x3fceacc4
D (862) event: created event loop 0x3fceacc4
V (872) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (872) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (882) intr_alloc: Connected src 38 to int 8 (cpu 0)
[I/USB] cherryusb, version: v1.5.0
[I/USB] ========== dwc2 hcd params ==========
[I/USB] CID:00000000
[I/USB] GSNPSID:4f54400a
[I/USB] GHWCFG1:00000000
[I/USB] GHWCFG2:224dd930
[I/USB] GHWCFG3:00c804b5
[I/USB] GHWCFG4:d3f0a030
[I/USB] dwc2 has 8 channels and dfifo depth(32-bit words) is 200
[I/usbh_hub] New full-speed device on Bus 0, Hub 1, Port 1 connected
[I/usbh_core] New device found,idVendor:0525,idProduct:a4a2,bcdDevice:0318
[I/usbh_core] The device has 2 bNumConfigurations
[I/usbh_core] The device has 2 interfaces
[I/usbh_core] Enumeration success, start loading class driver
[E/usbh_core] do not support Class:0x02,Subclass:0x02,Protocl:0xff
[E/usbh_core] do not support Class:0x0a,Subclass:0x00,Protocl:0x00

When I connect the device to my linux PC:
[1831464.123576] usb 1-1.5.3: new high-speed USB device number 17 using ehci-pci
[1831464.203375] usb 1-1.5.3: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 3.18
[1831464.203386] usb 1-1.5.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[1831464.203391] usb 1-1.5.3: Product: RNDIS/Ethernet Gadget
[1831464.203394] usb 1-1.5.3: Manufacturer: Linux 2.6.39+ with atmel_usba_udc
[1831464.210025] cdc_ether 1-1.5.3:1.0 eth1: register 'cdc_ether' at usb-0000:00:1a.0-1.5.3, CDC Ethernet Device, 4a:88:7b:11:a4:33
p

sakumisu, 2025-06-26:
From your log, your device has two bNumConfigurations,and linux select the two, but cherryusb select the one, you need to change with the two.

sakumisu, 2025-06-26:
change config_index to 1 in usbh_core.c

sakumisu, 2025-06-27:
How do you know 0xff is rndis? Maybe ncm has some tricks, i now use ch579 for cdc ncm.

sakumisu, 2025-06-27:
OK, i see it from linux code .

kcsct, 2025-06-27:
So any ideas where I can start looking to solve my issue?

kcsct, 2025-06-27:
I enabled debug in f_ncm.c gadget driver and from cherryusb I print the transmit frame

#f_ncm outout
reset ncm control 0
init ncm ctrl 0
notify speed 9728000
Notification 2a sent
notify connect false
Notification 00 sent
Host asked NTB parameters
ncm reqa1.80 v0000 i0000 l28
init ncm
activate ncm
ncm_open
notify speed 9728000
Notification 2a sent
notify speed 9728000
IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready
Notification 2a sent
notify connect true
Notification 00 sent
#no output further

# from static esp_err_t usbh_cdc_ncm_transmit(void *h, void *buffer, size_t len)
# I call usb_hexdump(buffer, len);
# the same data is being sent over and over
00000000:  FF FF FF FF FF FF 62 13   EF 44 04 49 08 00 45 00  ......b..D.I..E.
00000010:  01 50 00 00 00 00 40 11   79 9E 00 00 00 00 FF FF  .P....@.y.......
00000020:  FF FF 00 44 00 43 01 3C   C1 62 01 01 06 00 85 81  ...D.C.<.b......
00000030:  8D B9 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
00000040:  00 00 00 00 00 00 62 13   EF 44 04 49 00 00 00 00  ......b..D.I....
00000050:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
00000060:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
00000070:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
00000080:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
00000090:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
000000a0:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
000000b0:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
000000c0:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
000000d0:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
000000e0:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
000000f0:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
00000100:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
00000110:  00 00 00 00 00 00 63 82   53 63 35 01 01 39 02 05  ......c.Sc5..9..
00000120:  DC 0C 09 65 73 70 72 65   73 73 69 66 37 04 01 03  ...espressif7...
00000130:  1C 06 3D 07 01 62 13 EF   44 04 49 FF 00 00 00 00  ..=..b..D.I.....
00000140:  00 00 00 00 00 00 00 00   00 00 00 00 00 00 00 00  ................
00000150:  00 00 00 00 00 00 00 00   00 00 00 00 00 00        ..............

sakumisu, 2025-06-27:
I have no idea, sorry.

sakumisu, 2025-06-27:
You can use ch397 for a try, we have worked with it.

kcsct, 2025-06-27:
Well its the linux gadget drivers, are they not standard?  They work on linux and windows hosts, both RNDIS and CDC_NCM.

sakumisu, 2025-06-27:
Linux does many patches for them, why you think they are standard?

sakumisu, 2025-06-27:
> OK, i see it from linux code .
> 
> static const struct [usb_device_id](https://elixir.bootlin.com/linux/v6.15.3/C/ident/usb_device_id) [products](https://elixir.bootlin.com/linux/v6.15.3/C/ident/products) [] = { { /* 2Wire HomePortal 1000SW _/ [USB_DEVICE_AND_INTERFACE_INFO](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_DEVICE_AND_INTERFACE_INFO)(0x1630, 0x0042, [USB_CLASS_COMM](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_CLASS_COMM), 2 /_ ACM _/, 0x0ff), .[driver_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/driver_info) = (unsigned long) &[rndis_poll_status_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/rndis_poll_status_info), }, { /_ Hytera Communications DMR radios' "Radio to PC Network" _/ [USB_VENDOR_AND_INTERFACE_INFO](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_VENDOR_AND_INTERFACE_INFO)(0x238b, [USB_CLASS_COMM](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_CLASS_COMM), 2 /_ ACM _/, 0x0ff), .[driver_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/driver_info) = (unsigned long)&[rndis_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/rndis_info), }, { /_ ZTE WWAN modules _/ [USB_VENDOR_AND_INTERFACE_INFO](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_VENDOR_AND_INTERFACE_INFO)(0x19d2, [USB_CLASS_WIRELESS_CONTROLLER](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_CLASS_WIRELESS_CONTROLLER), 1, 3), .[driver_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/driver_info) = (unsigned long)&[zte_rndis_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/zte_rndis_info), }, { /_ ZTE WWAN modules, ACM flavour _/ [USB_VENDOR_AND_INTERFACE_INFO](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_VENDOR_AND_INTERFACE_INFO)(0x19d2, [USB_CLASS_COMM](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_CLASS_COMM), 2 /_ ACM _/, 0x0ff), .[driver_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/driver_info) = (unsigned long)&[zte_rndis_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/zte_rndis_info), }, { /_ RNDIS is MSFT's un-official variant of CDC ACM _/ [USB_INTERFACE_INFO](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_INTERFACE_INFO)([USB_CLASS_COMM](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_CLASS_COMM), 2 /_ ACM _/, 0x0ff), .[driver_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/driver_info) = (unsigned long) &[rndis_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/rndis_info), }, { /_ "ActiveSync" is an undocumented variant of RNDIS, used in WM5 _/ [USB_INTERFACE_INFO](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_INTERFACE_INFO)([USB_CLASS_MISC](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_CLASS_MISC), 1, 1), .[driver_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/driver_info) = (unsigned long) &[rndis_poll_status_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/rndis_poll_status_info), }, { /_ RNDIS for tethering _/ [USB_INTERFACE_INFO](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_INTERFACE_INFO)([USB_CLASS_WIRELESS_CONTROLLER](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_CLASS_WIRELESS_CONTROLLER), 1, 3), .[driver_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/driver_info) = (unsigned long) &[rndis_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/rndis_info), }, { /_ Novatel Verizon USB730L */ [USB_INTERFACE_INFO](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_INTERFACE_INFO)([USB_CLASS_MISC](https://elixir.bootlin.com/linux/v6.15.3/C/ident/USB_CLASS_MISC), 4, 1), .[driver_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/driver_info) = (unsigned long) &[rndis_info](https://elixir.bootlin.com/linux/v6.15.3/C/ident/rndis_info), }, { }, // END };

Look this, without these driver match,  do you think your rndis can work fine?

closed 2025-06-27 by sakumisu