cherry-embedded/CherryUSB #336: Enumeration failed with CONFIG_USBHOST_GET_STRING_DESC when HS hub is connected.

AndyDevLat, 2025-07-18:
### 描述一下这个bug / Describe the bug

With HS hub you cannot read any string descriprtor via usbh_get_string_desc API.
For test I have enabled CONFIG_USBHOST_GET_STRING_DESC and it failed up too even with just hub - nothing connected to it.
If you disable CONFIG_USBHOST_GET_STRING_DESC then hub enumaration is ok, but if you try usbh_get_string_desc on some device it will fail too.

### 复现步骤 / To Reproduce

Try to attach HS hub with CONFIG_USBHOST_GET_STRING_DESC defined.

### 设备信息 / Target Device

ESP32-P4-Function-EV-Board with CH334R hub connected to HS port of ESP32-P4 (0x50000000).

### 日志 / Log

I/usbh_hub] New high-speed device on Bus 0, Hub 1, Port 1 connected
[I/usbh_core] New device found,idVendor:1a86,idProduct:8091,bcdDevice:1310
[I/usbh_core] The device has 1 bNumConfigurations
[I/usbh_core] The device has 1 interfaces
[E/usbh_core] Failed to get Manufacturer string,errorcode:-8
[E/usbh_hub] Port 1 enumerate fail

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2025-07-18:
PC tolerates that some commands are not supported, but we indicate here that they must be supported

AndyDevLat, 2025-07-18:
Yes but device supports it and it still fails with hub (when i remove CONFIG_USBHOST_GET_STRING_DESC and call this API inside my  .connect callback). If i remove hub and attach device dirctly then  usbh_get_string_desc works. Most likely some issue with split transactions.

AndyDevLat, 2025-07-18:
CH334R is HS hub which supports split transactions for any FS device connected to its downstream port (any HS hub has transaction translator according to standard). 
Do you mean the hub itself is buggy?

sakumisu, 2025-07-18:
I said hs hub itself does not support string cmd, please do not connect others device on hub. we are talking about hub device with -8 error, not device on hub.

AndyDevLat, 2025-07-18:
Oh, i see. Will try some other hub then and let you know ...

AndyDevLat, 2025-07-18:
I found other hub:  CY7C65632 (from CY4608 HX2VL Development Kit). I see it supports usbh_get_string_desc for USB_STRING_MFC_INDEX, but stalls on USB_STRING_PRODUCT_INDEX. To make tests easier i set CONFIG_USBHOST_GET_STRING_DESC, but commented all code for USB_STRING_PRODUCT_INDEX and USB_STRING_SERIAL_INDEX inside usb_core.c.
With just hub connected i have this log: 

[I/usbh_hub] New high-speed device on Bus 0, Hub 1, Port 1 connected
[I/usbh_core] New device found,idVendor:04b4,idProduct:6570,bcdDevice:3299
[I/usbh_core] The device has 1 bNumConfigurations
[I/usbh_core] The device has 1 interfaces
[I/usbh_core] Manufacturer: USB2.0 Hub
[I/usbh_core] Enumeration success, start loading class driver
[I/usbh_core] Loading hub class driver
[I/usbh_hub] Ep=81 Attr=03 Mps=1 Interval=12 Mult=00
[I/usbh_hub] Register HUB Class:/dev/hub2

Now, i have tried to connect DualShock 4 to one port and got this:

[I/usbh_hub] New full-speed device on Bus 0, Hub 2, Port 4 connected
[I/usbh_core] New device found,idVendor:054c,idProduct:09cc,bcdDevice:0100
[I/usbh_core] The device has 1 bNumConfigurations
[I/usbh_core] The device has 4 interfaces
[E/usbh_core] Failed to get Manufacturer string,errorcode:-11
[E/usbh_hub] Port 4 enumerate fail

But it supports this command for sure.

Then i reset the board and tried other controller (Nintendo Switch Pro), and i got this:
I/usbh_hub] New full-speed device on Bus 0, Hub 2, Port 4 connected
[I/usbh_core] New device found,idVendor:057e,idProduct:2009,bcdDevice:0210
[I/usbh_core] The device has 1 bNumConfigurations
[I/usbh_core] The device has 1 interfaces
It hanged when trying to get manufacturer string and then later i got:
Guru Meditation Error: Core  1 panic'ed (Load access fault). Exception was unhandled

So something weird is going on.

If hub is not used all devices work fine.
For example, on Nintendo i get:
[I/usbh_core] New device found,idVendor:057e,idProduct:2009,bcdDevice:0210
[I/usbh_core] The device has 1 bNumConfigurations
[I/usbh_core] The device has 1 interfaces
[I/usbh_core] Manufacturer: Nintendo Co., Ltd.
[I/usbh_core] Enumeration success, start loading class driver
...

sakumisu, 2025-07-18:
I don't know this, you can disable hub.

AndyDevLat, 2025-07-18:
Yes, maybe it is ESP32-P4 specific.

sakumisu, 2025-07-18:
I can reproduce issue when fs device connects with hs hub(nothing problem when disable get desc), i will check reason.

sakumisu, 2025-07-18:
This issue is about hs hub itself, not fs device on hub, so i close it.

closed 2025-07-18 by sakumisu

AndyDevLat, 2025-07-18:
Hi!

Thank you - Nintendo Switch Pro now working on HS hub.
But some devices still have problem, on DualShock 4 i get this:

I/usbh_hub] New full-speed device on Bus 0, Hub 2, Port 4 connected
[I/usbh_core] New device found,idVendor:054c,idProduct:09cc,bcdDevice:0100
[I/usbh_core] The device has 1 bNumConfigurations
[I/usbh_core] The device has 4 interfaces
[I/usbh_core] Enumeration success, start loading class driver
[E/usbh_core] do not support Class:0x01,Subclass:0x01,Protocl:0x00
[E/usbh_core] do not support Class:0x00,Subclass:0x00,Protocl:0x00
[E/usbh_core] do not support Class:0x00,Subclass:0x00,Protocl:0x00
[E/usbh_core] do not support Class:0x00,Subclass:0x00,Protocl:0x00

Looks like config descriptor is corrupt.
When attached directly it works fine, so seems still some issue with split control transfers ....

Regards,

Andrew

Update with latest commit will fix this.

sakumisu, 2025-07-18:
This maybe cache issue not usb.

sakumisu, 2025-07-18:
> You cannot disable cache on ESP32-P4. If i comment CONFIG_USB_DCACHE_ENABLE it stops working at all, no device descriptor can be read at all,
> 
> I/usbh_hub] New full-speed device on Bus 0, Hub 1, Port 1 connected [E/usbh_core] Failed to get device descriptor,errorcode:-8 [E/usbh_hub] Port 1 enumerate fail

Sorry, you cannot comment CONFIG_USB_DCACHE_ENABLE, i mean you disable dcache in esp not usb. like esp_dcache_disable(), i don't know if this api has.

AndyDevLat, 2025-07-18:
>try this https://github.com/cherry-embedded/CherryUSB/commit/cc3e91e8d70693b0dc0e954d5c709b15f0e06c6d

Yes! With this commit it is working now! Will do more tests with other devices and let you know if some problems.

AndyDevLat, 2025-07-18:
What star you mean?

sakumisu, 2025-07-18:
ESP official is working on cherryusb in esp-idf, these will merge in esp component registry.

sakumisu, 2025-07-18:
> What star you mean?

here.

[image]

AndyDevLat, 2025-07-18:
Btw, i have one small request: can you make setting for CONFIG_USBHOST_PSC_AFFINITY?
It is very useful on dual core system for more fine managent on which core the hub task is running and simpflies work if we need to synchronize some other tasks with it.
Of course i can easily edit usb_osal_idf.c, but out-of-the-box feature would be nice.

AndyDevLat, 2025-07-18:
>we will implement scatter dma mode in the future but it is in commercial charge.

Espressif guys said that in this mode split transactions do not work - it is hardware limitation of  DWC2 (at least on ESP32-P4). The are supported in Buffer DMA or in Slave modes only.
So, if we need FS devices on HS hub, Scatter Gather DMA is absolutely not interesting for us.

sakumisu, 2025-07-18:
> Btw, i have one small request: can you make setting for CONFIG_USBHOST_PSC_AFFINITY? It is very useful on dual core system for more fine managent on which core the hub task is running and simpflies [...]

you can create new issue for this.