cherry-embedded/CherryUSB #349: duplicate get

linuxhj, 2025-08-01:
### 描述一下这个bug / Describe the bug

usbh_core.c
`557      USB_LOG_ERR("Failed to get get SerialNumber string,errorcode:%d\r\n", ret);`
duplicate get

### 复现步骤 / To Reproduce

None

### 设备信息 / Target Device

None

### 日志 / Log

[D/usbh_core] Setup: bmRequestType 0x80, bRequest 0x06, wValue 0x0303, wIndex 0
[E/usbh_core] Failed to **get get** SerialNumber string,errorcode:-8
[E/usbh_hub] Port 1 enumerate fail

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2025-08-01:
Will fix, btw, -8 mean your device does not support serial number, you need to ignore this error. Currently usb device must support these three string desc in cherryusb.

linuxhj, 2025-08-04:
> Will fix, btw, -8 mean your device does not support serial number, you need to ignore this error. Currently usb device must support these three string desc in cherryusb.

Thanks a lot