cherry-embedded/CherryUSB #354: STM32新FSDEV的支持

rayc345, 2025-08-20:
### 描述一下这个bug / Describe the bug

目前FSDEV的驱动支持，不支持部分新的型号，如G0 H5 U535/545 U3 U0。这些型号中的FSDEV IP有2048 Bytes的PMA，而且必须使用字(4 Bytes)访问PMA。
此外CNTR, ISTR等寄存器也从16位拓展到了32位，拓展的16位主要是和主机模式以及iso传输相关。
我对FSDEV进行了拓展，更改的内容主要有：
1.删除了usb_fsdev_reg.h中的寄存器定义，改为引用stm32g0xx.h这种CMSIS定义。
2.使用LL库中的PMA访问API替代目前自定义的API。

由于在usb_fsdev_reg.h中直接使用ST的驱动，这里的移植性下降了，所以我的想法是把修改后的FSDEV驱动作为一个新的port，例如命名为FSDEV2，放在port文件夹下，专门用于STM32的驱动？
想把修改合并到主线，问下作者觉得这么修改合适吗？有没有其他的建议，我自己处理下？

### 复现步骤 / To Reproduce

无

### 设备信息 / Target Device

我手头有的板子：STM32U385KG
FSDEV IP

### 日志 / Log

无

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2025-08-20:
fsdev不做维护，不可添加到主线

sakumisu, 2025-08-20:
我没有义务去支持st的所有芯片，并且去支持一个不带主机的ip

rayc345, 2025-08-20:
好的。我看仓库和说明网页上都没有对fsdev的停止维护做出说明，麻烦作者后面在仓库的readme备注下各个ip的支持情况，以便使用者了解，然后自行解决需求，谢谢。

sakumisu, 2025-08-20:
额，这个加群就知道了。并且有且只有 FSDEV 不维护哈

closed 2025-08-20 by sakumisu