cherry-embedded/CherryUSB #355: demo audio_v1_mic_speaker_multichan_template 使用printf打印log，device端进入异常

yueyue30, 2025-08-20:
### 描述一下这个bug / Describe the bug

1. 环境：freertos, libc库
2. 使用Cherryusb demo中audio_v1_mic_speaker_multichan_template例程
3. 例程中usbd_audio_open和usbd_audio_close 打印log使用了printf
4. 基于以上环境，当device在open/close接口使用printf（带锁）打印log，device会异常
5. 像其他demo中一样使用USB_LOG_RAW打印信息则功能正常

### 复现步骤 / To Reproduce

跑demo例程audio_v1_mic_speaker_multichan_template

### 设备信息 / Target Device

USB IP: MUSB

### 日志 / Log

=== CherryUSB Example ===
=== Version:v1.0.0 ===
usb_dc_low_level_init.
CLCLOSE2
OSE2

### 商业 / Business

- [ ] 是 / Yes

sakumisu, 2025-08-20:
你printf实现问题，自行解决

sakumisu, 2025-08-20:
USB_LOG_RAW就是调用printf，请你点进去看看谢谢

yueyue30, 2025-08-20:
没错，我自己工程里CONFIG_USB_PRINTF是自己定义的，如果CONFIG_USB_PRINTF直接使用printf, 而我的系统中直接使用libc 中的printf，demo进入中断回调直接用printf打印带锁，device就会异常。所以CONFIG_USB_PRINTF会自定义实现。

sakumisu, 2025-08-20:
后面再看看还有哪些漏了一并修改

sakumisu, 2025-08-20:
重新打开，等后面修完了再关闭，谢谢提醒

yueyue30, 2025-08-20:
不客气

sakumisu, 2025-08-27:
done

closed 2025-08-27 by sakumisu