cherry-embedded/CherryUSB #360: USB host modem - Interface num 32 overflow

pioneerAlone, 2025-09-09:
rtthread 5.1.0 + cherryusb 1.5.2 + STM32F429BIT6 ，配置Host mode 并使能CDC ACM driver, 

 \ | /
- RT -     Thread Operating System
 / | \     5.1.0 build Sep  9 2025 17:12:12
 2006 - 2024 Copyright by RT-Thread team
[D/main] Hello RT-Thread!
[I/USB] ========== dwc2 hcd params ==========
[I/USB] CID:00001100
[I/USB] GSNPSID:4f54281a
[I/USB] GHWCFG1:00000000
[I/USB] GHWCFG2:229ed590
[I/USB] GHWCFG3:03f403e8
[I/USB] GHWCFG4:17f00030
[I/USB] dwc2 has 12 channels and dfifo depth(32-bit words) is 1012
msh >
// 插入USB 设备后显示如下,无论插入什么USB 设备都是这样，且我确认了USB device 只有8 个interface, 亲提供一些排查建议
[I/usbh_hub] New full-speed device on Bus 0, Hub 1, Port 1 connected
[I/usbh_core] New device found,idVendor:2c7c,idProduct:0700,bcdDevice:0000
[I/usbh_core] The device has 1 bNumConfigurations
[E/usbh_core] Interface num 32 overflow
[E/usbh_core] Parse config descriptor fail
[E/usbh_hub] Port 1 enumerate fail

[image]

sakumisu, 2025-09-09:
你错了，interface num不代表是从0开始

sakumisu, 2025-09-09:
举例，你描述符里面接口0的 interface num 是31，那么接口1是32，则直接溢出

pioneerAlone, 2025-09-09:
谢谢！

我把以下宏改为36 ，可以枚举了，
#define CONFIG_USBHOST_MAX_INTERFACES       36

通过下面命令查看 interface num 是从0开始的呀。
`lsusb -v

Bus 0, Hub 1, Port 1, dev addr:0x02, VID:PID 0x2c7c:0x0700
Device Descriptor:
bLength: 0x12
bDescriptorType: 0x01
bcdUSB: 0x0200
bDeviceClass: 0x00
bDeviceSubClass: 0x00
bDeviceProtocol: 0x00
bMaxPacketSize0: 0x40
idVendor: 0x2c7c
idProduct: 0x0700
bcdDevice: 0x0000
iManufacturer: 0x03
iProduct: 0x02
iSerialNumber: 0x04
bNumConfigurations: 0x01
Config Descriptor:
bLength: 0x09
bDescriptorType: 0x02
wTotalLength: 0x0073
bNumInterfaces: 0x04
bConfigurationValue: 0x01
iConfiguration: 0x01
bmAttributes: 0xe0
bMaxPower: 0xfa
        Interface Descriptor:
        bLength: 0x09
        bDescriptorType: 0x04
        **bInterfaceNumber: 0x00**
        bAlternateSetting: 0x00
        bNumEndpoints: 0x02
        bInterfaceClass: 0xff
        bInterfaceSubClass: 0xff
        bInterfaceProtocol: 0xff
        iInterface: 0x00
                Endpoint Descriptor:
                bLength: 0x07
                bDescriptorType: 0x05
                bEndpointAddress: 0x81
                bmAttributes: 0x02
                wMaxPacketSize: 0x0040
                bInterval: 0x00
                Endpoint Descriptor:
                bLength: 0x07
                bDescriptorType: 0x05
                bEndpointAddress: 0x01
                bmAttributes: 0x02
                wMaxPacketSize: 0x0040
                bInterval: 0x00
        Interface Descriptor:
        bLength: 0x09
        bDescriptorType: 0x04
        bInterfaceNumber: 0x02
        bAlternateSetting: 0x00
        bNumEndpoints: 0x03
        bInterfaceClass: 0xff
        bInterfaceSubClass: 0xff
        bInterfaceProtocol: 0xff
        iInterface: 0x00
                Endpoint Descriptor:
                bLength: 0x07
                bDescriptorType: 0x05
                bEndpointAddress: 0x83
                bmAttributes: 0x03
                wMaxPacketSize: 0x0040
                bInterval: 0x05
                Endpoint Descriptor:
                bLength: 0x07
                bDescriptorType: 0x05
                bEndpointAddress: 0x84
                bmAttributes: 0x02
                wMaxPacketSize: 0x0040
                bInterval: 0x00
                Endpoint Descriptor:
                bLength: 0x07
                bDescriptorType: 0x05
                bEndpointAddress: 0x03
                bmAttributes: 0x02
                wMaxPacketSize: 0x0040
                bInterval: 0x00
        Interface Descriptor:
        bLength: 0x09
        bDescriptorType: 0x04
        bInterfaceNumber: 0x03
        bAlternateSetting: 0x00
        bNumEndpoints: 0x03
        bInterfaceClass: 0xff
        bInterfaceSubClass: 0xff
        bInterfaceProtocol: 0xff
        iInterface: 0x00
                Endpoint Descriptor:
                bLength: 0x07
                bDescriptorType: 0x05
                bEndpointAddress: 0x85
                bmAttributes: 0x03
                wMaxPacketSize: 0x0040
                bInterval: 0x05
                Endpoint Descriptor:
                bLength: 0x07
                bDescriptorType: 0x05
                bEndpointAddress: 0x86
                bmAttributes: 0x02
                wMaxPacketSize: 0x0040
                bInterval: 0x00
                Endpoint Descriptor:
                bLength: 0x07
                bDescriptorType: 0x05
                bEndpointAddress: 0x04
                bmAttributes: 0x02
                wMaxPacketSize: 0x0040
                bInterval: 0x00
`

另一个问题是

`[D/main] Hello RT-Thread!
[I/USB] ========== dwc2 hcd params ==========
[I/USB] CID:00001100
[I/USB] GSNPSID:4f54281a
[I/USB] GHWCFG1:00000000
[I/USB] GHWCFG2:229ed590
[I/USB] GHWCFG3:03f403e8
[I/USB] GHWCFG4:17f00030
[I/USB] dwc2 has 12 channels and dfifo depth(32-bit words) is 1012
msh >[I/usbh_hub] New full-speed device on Bus 0, Hub 1, Port 1 connected
[I/usbh_core] New device found,idVendor:2c7c,idProduct:0700,bcdDevice:0000
[I/usbh_core] The device has 1 bNumConfigurations
[I/usbh_core] The device has 4 interfaces
[I/usbh_core] Enumeration success, start loading class driver

// 这里加载驱动失败，是以为没有识别可用的驱动所致是吗？
[E/usbh_core] do not support Class:0xff,Subclass:0xff,Protocl:0xff
[E/usbh_core] do not support Class:0x00,Subclass:0x00,Protocl:0x00
[E/usbh_core] do not support Class:0xff,Subclass:0xff,Protocl:0xff
[E/usbh_core] do not support Class:0xff,Subclass:0xff,Protocl:0xff
`

sakumisu, 2025-09-09:
肯定啊，厂家自定义驱动，你要编写厂家驱动啊，搞的你电脑不装驱动一样。其次，你改成36是不要ram了吗，建议在解析描述符中加断点去看为什么解析错误。

pioneerAlone, 2025-09-09:
好的，感谢！